jneilliii/OctoPrint-BGCode #1: Additional instructions needed for manual build of libbgcode

b-morgan, 2023-11-24:
In [building.md](https://github.com/prusa3d/libbgcode/blob/main/doc/building.md) it says:

> where \<install-dir\> is an arbitrary install folder.

and I'm guessing you are expecting the library to be somewhere so your instructions show say where that is.

jneilliii, 2023-11-24:
Yeah, until I'm able to test this on a Linux machine I'm not going to say, but in Windows I ran a few different options until it actually compiled without errors. Maybe instructions should be better defined on the dependency repo too.

jneilliii, 2023-11-24:
I think I got the auto-install to work (at least with octopi images) and have updated the readme with additional details.

b-morgan, 2023-11-24:
On my RPi 4B with a virtual printer I was able to upload and "print" a .bgcode file so I believe you can close this issue.

closed 2023-11-25 by jneilliii