jneilliii/OctoPrint-BGCode #3: Upload of bgcode fails with utf-8 error

oxivanisher, 2023-11-24:
First: I love that you take on this plugin!

I installed it in my octopi installation and the setup itself ran without any problem. Sure, it took some time but this is expected since the lib has to be built. But when I try to upload my .bgcode, I get a error:

```plaintext
==> /home/pi/.octoprint/logs/octoprint.log <==
2023-11-24 22:32:44,337 - octoprint.plugins.SlicerEstimator - INFO - Detected PrusaSlicer
2023-11-24 22:32:44,347 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/util/flask.py", line 1613, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/vendor/flask_principal.py", line 196, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/api/files.py", line 671, in uploadGcodeFile
    added_file = fileManager.add_file(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintTimeGenius/__init__.py", line 580, in new_add_file
    return self._file_manager.original_add_file(destination, path, file_object, links, allow_overwrite, printer_profile, None, display)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/__init__.py", line 727, in add_file
    path_in_storage = self._storage(destination).add_file(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/storage.py", line 894, in add_file
    file_object.save(file_path)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/util.py", line 102, in save
    shutil.copyfileobj(source, dest)
  File "/usr/lib/python3.9/shutil.py", line 205, in copyfileobj
    buf = fsrc_read(length)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/util.py", line 211, in read
    line = self.input_stream.readline()
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/util.py", line 230, in readinto
    read = self.read(n)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/filemanager/util.py", line 214, in read
    processed_line = self.process_line(line)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_SlicerEstimator/metadata.py", line 160, in process_line
    decoded_line = line.decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8a in position 1: invalid start byte
```

My setup:
* OctoPi 1.0.0
* Raspbian GNU/Linux 11 (bullseye)
* All OS packages are up to date
* Raspberry Pi 3 Model B Plus Rev 1.3 (1GB RAM)
* Printer PrusaMK4 with FW 5.1
* pybcode version 0.2.0 was installed automatically in the octoprint venv (/home/pi/oprint)

I tested this with several .bgcode files (PrusaSlicer 2.7.0) and all fail the same way. Here is the simplest one:
[Shape-Box_0.6n_0.25mm_PLA_MK4_26m.zip](https://github.com/jneilliii/OctoPrint-BGCode/files/13462203/Shape-Box_0.6n_0.25mm_PLA_MK4_26m.zip)

If I am early with testing please let me know, my aim is to support you, not to get on your nerves. ;)

jneilliii, 2023-11-24:
Does it happen with Slicer Estimator and Print Time Genius plugin's disabled?

oxivanisher, 2023-11-24:
* Disabling only Slicer Estimator: Upload works
* Disabling only Print Time Genius: Upload fails
* Disabling both: Upload works (obviously ^^)

So Slicer Estimator is the culprit.

NilsRo, 2023-11-26:
Hi. I am the maintainer of SlicerEstimator...I am not using Prusa and did not find many about the format with a short Google Search...so perhaps you can help out:

- What is that special with that format? Is it binary as the UTF-8 reader could not read it as there is a error in the file itself?
- At this step of conversion is the file in bgcode or standard gcode format?

Could somebody provide examples? As other plugins shows problems could this the same problem?

Cheers
Nils

jneilliii, 2023-11-26:
I think I may have figured it out NilsRo. Because we are both using file preprocessors I think yours is trying to run first on a binary file. I'm about to test a change to this plugin to verify that is the case. If so then we'll be able to close both these tickets.

oxivanisher, 2023-11-26:
I can tell something about the [new format](https://github.com/prusa3d/libbgcode). It is kind of a binary representation of the classic ASCII format. You can convert files both ways without loss and the basic idea of Prusa ist to optimize file size and with that speed for network uploads and such. With the release of Prusaslicer 2.7 (just last week) they set bgcode as default format for the mk4, mini and xl.
But this new format is pretty young and the specification is on github, but right now there are [no releases in pypi](https://github.com/prusa3d/libbgcode/issues/31) of this lib.
This plugin just takes the bgcode file and converts it to gcode.

EDIT: I was writing this when @jneilliii already replied. So ignore this.

oxivanisher, 2023-11-26:
That did the trick. thanks!
Everything is working as expected. 👍