jneilliii/OctoPrint-BGCode #5: Install failes

fly74, 2023-11-25:
Hi Dev,

I want to give it a try, but:

```
2023-11-25 18:42:11,382   /usr/bin/python3.11 -m pip --disable-pip-version-check install file:///tmp/tmpsmhenv5z/OctoPrint-BGCode-master.zip --no-cache-dir --user
2023-11-25 18:42:12,690 > Processing /tmp/tmpsmhenv5z/OctoPrint-BGCode-master.zip
2023-11-25 18:42:12,690 > Preparing metadata (setup.py): started
2023-11-25 18:42:14,294 > Preparing metadata (setup.py): finished with status 'done'
2023-11-25 18:42:14,294 > Collecting pybgcode@ https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
2023-11-25 18:42:14,294 > Downloading https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
2023-11-25 18:42:14,294 > | 474.1 kB 2.0 MB/s 0:00:00
2023-11-25 18:42:14,294 > Installing build dependencies: started
2023-11-25 18:42:18,486 > Installing build dependencies: finished with status 'done'
2023-11-25 18:42:18,486 > Getting requirements to build wheel: started
2023-11-25 18:42:18,486 > Getting requirements to build wheel: finished with status 'done'
2023-11-25 18:42:18,486 > Installing backend dependencies: started
2023-11-25 18:42:22,854 > Installing backend dependencies: finished with status 'done'
2023-11-25 18:42:22,854 > Preparing metadata (pyproject.toml): started
2023-11-25 18:43:12,863 ! error: subprocess-exited-with-error
2023-11-25 18:43:12,863 ! 
2023-11-25 18:43:12,864 ! × Preparing metadata (pyproject.toml) did not run successfully.
2023-11-25 18:43:12,864 ! │ exit code: 1
2023-11-25 18:43:12,864 ! ╰─> [92 lines of output]
2023-11-25 18:43:12,864 ! -- The C compiler identification is GNU 7.5.0
2023-11-25 18:43:12,864 ! -- The CXX compiler identification is GNU 7.5.0
2023-11-25 18:43:12,864 ! -- Detecting C compiler ABI info
2023-11-25 18:43:12,864 ! -- Detecting C compiler ABI info - done
2023-11-25 18:43:12,864 ! -- Check for working C compiler: /usr/bin/cc - skipped
2023-11-25 18:43:12,864 ! -- Detecting C compile features
2023-11-25 18:43:12,864 ! -- Detecting C compile features - done
2023-11-25 18:43:12,864 ! -- Detecting CXX compiler ABI info
2023-11-25 18:43:12,864 ! -- Detecting CXX compiler ABI info - done
2023-11-25 18:43:12,864 ! -- Check for working CXX compiler: /usr/bin/c++ - skipped
2023-11-25 18:43:12,864 ! -- Detecting CXX compile features
2023-11-25 18:43:12,864 ! -- Detecting CXX compile features - done
2023-11-25 18:43:12,865 ! -- Building the dependencies with preset python-module
2023-11-25 18:43:12,865 ! -- build dir = /tmp/pip-install-vwr4vu7d/pybgcode_4221e2ff39a74383b33b4bcc085b7935/deps/build-python-module
2023-11-25 18:43:12,865 ! -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
2023-11-25 18:43:12,865 ! -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
2023-11-25 18:43:12,865 ! -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
2023-11-25 18:43:12,865 ! -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
2023-11-25 18:43:12,865 ! -- Performing Test COMPILER_HAS_DEPRECATED_ATTR
2023-11-25 18:43:12,865 ! -- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
2023-11-25 18:43:12,865 ! -- Found ZLIB: /usr/lib64/libz.so (found suitable version "1.2.13", minimum required is "1.0")
2023-11-25 18:43:12,865 ! -- Found Boost: /tmp/pip-install-vwr4vu7d/pybgcode_4221e2ff39a74383b33b4bcc085b7935/deps/build-python-module/destdir/usr/local/lib64/cmake/Boost-1.82.0/BoostConfig.cmake (found suitable version "1.82.0", minimum required is "1.78")
2023-11-25 18:43:12,865 ! -- Found Python3: /usr/bin/python3.11 (found version "3.11.5") found components: Interpreter Development Development.Module Development.Embed
2023-11-25 18:43:12,865 ! -- Performing Test HAS_FLTO
2023-11-25 18:43:12,865 ! -- Performing Test HAS_FLTO - Success
2023-11-25 18:43:12,865 ! -- Found pybind11: /tmp/pip-install-vwr4vu7d/pybgcode_4221e2ff39a74383b33b4bcc085b7935/deps/build-python-module/destdir/usr/local/include (found version "2.11.1")
2023-11-25 18:43:12,865 ! -- Found Boost: /tmp/pip-install-vwr4vu7d/pybgcode_4221e2ff39a74383b33b4bcc085b7935/deps/build-python-module/destdir/usr/local/lib64/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0") found components: nowide
2023-11-25 18:43:12,865 ! -- Configuring done (43.0s)
2023-11-25 18:43:12,865 ! -- Generating done (0.0s)
2023-11-25 18:43:12,866 ! CMake Warning:
2023-11-25 18:43:12,866 ! Manually-specified variables were not used by the project:
2023-11-25 18:43:12,866 ! 
2023-11-25 18:43:12,866 ! PY_BUILD_CMAKE_PACKAGE_NAME
2023-11-25 18:43:12,866 ! 
2023-11-25 18:43:12,866 ! 
2023-11-25 18:43:12,866 ! -- Build files have been written to: /tmp/pip-install-vwr4vu7d/pybgcode_4221e2ff39a74383b33b4bcc085b7935/.py-build-cmake_cache/cp311-cp311-linux_x86_64
2023-11-25 18:43:12,866 ! [ 11%] Building CXX object upstream/src/LibBGCode/core/CMakeFiles/bgcode_core.dir/core.cpp.o
2023-11-25 18:43:12,866 ! [ 22%] Linking CXX static library libbgcode_core.a
2023-11-25 18:43:12,866 ! [ 22%] Built target bgcode_core
2023-11-25 18:43:12,866 ! [ 33%] Building CXX object upstream/src/LibBGCode/binarize/CMakeFiles/bgcode_binarize.dir/binarize.cpp.o
2023-11-25 18:43:12,866 ! [ 44%] Building CXX object upstream/src/LibBGCode/binarize/CMakeFiles/bgcode_binarize.dir/meatpack.cpp.o
2023-11-25 18:43:12,866 ! [ 55%] Linking CXX static library libbgcode_binarize.a
2023-11-25 18:43:12,866 ! [ 55%] Built target bgcode_binarize
2023-11-25 18:43:12,867 ! [ 66%] Building CXX object upstream/src/LibBGCode/convert/CMakeFiles/bgcode_convert.dir/convert.cpp.o
2023-11-25 18:43:12,868 ! /tmp/pip-install-vwr4vu7d/pybgcode_4221e2ff39a74383b33b4bcc085b7935/src/LibBGCode/convert/convert.cpp: In function ‘void bgcode::convert::to_int(const string_view&, Integer&)’:
2023-11-25 18:43:12,868 ! /tmp/pip-install-vwr4vu7d/pybgcode_4221e2ff39a74383b33b4bcc085b7935/src/LibBGCode/convert/convert.cpp:177:16: error: ‘from_chars_result’ in namespace ‘std’ does not name a type
[... 34 lines cut ...]
2023-11-25 18:43:12,870 ! cmaker.build()
2023-11-25 18:43:12,870 ! File "/tmp/pip-build-env-p9knntzv/overlay/lib/python3.11/site-packages/py_build_cmake/cmake.py", line 190, in build
2023-11-25 18:43:12,870 ! self.run(cmd, cwd=cwd, check=True, env=env)
2023-11-25 18:43:12,870 ! File "/tmp/pip-build-env-p9knntzv/overlay/lib/python3.11/site-packages/py_build_cmake/cmake.py", line 68, in run
2023-11-25 18:43:12,870 ! return self.runner.run(*args, **kwargs)
2023-11-25 18:43:12,870 ! ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-11-25 18:43:12,870 ! File "/tmp/pip-build-env-p9knntzv/overlay/lib/python3.11/site-packages/py_build_cmake/cmd_runner.py", line 25, in run
2023-11-25 18:43:12,870 ! return sp_run(*args, **kwargs)
2023-11-25 18:43:12,870 ! ^^^^^^^^^^^^^^^^^^^^^^^
2023-11-25 18:43:12,870 ! File "/usr/lib64/python3.11/subprocess.py", line 571, in run
2023-11-25 18:43:12,870 ! raise CalledProcessError(retcode, process.args,
2023-11-25 18:43:12,870 ! subprocess.CalledProcessError: Command '['cmake', '--build', '/tmp/pip-install-vwr4vu7d/pybgcode_4221e2ff39a74383b33b4bcc085b7935/.py-build-cmake_cache/cp311-cp311-linux_x86_64', '--config', 'Release', '--target', 'build_and_test']' returned non-zero exit status 2.
2023-11-25 18:43:12,870 ! [end of output]
2023-11-25 18:43:12,871 ! 
2023-11-25 18:43:12,871 ! note: This error originates from a subprocess, and is likely not a problem with pip.
2023-11-25 18:43:12,871 ! error: metadata-generation-failed
2023-11-25 18:43:12,871 ! 
2023-11-25 18:43:12,871 ! × Encountered error while generating package metadata.
2023-11-25 18:43:12,871 ! ╰─> See above for output.
2023-11-25 18:43:12,871 ! 
2023-11-25 18:43:12,871 ! note: This is an issue with the package mentioned above, not pip.
2023-11-25 18:43:12,871 ! hint: See above for details.
2023-11-25 18:43:13,372 > Preparing metadata (pyproject.toml): finished with status 'error'

```

jneilliii, 2023-11-25:
This appears to be an upstream issue with libbgcode. Not sure if that's a python 3.11 issue or you don't have all the necessary build tools in your environment. I recommend you see if you can build that outside of OctoPrint.

fly74, 2023-11-25:
> I recommend you see if you can build that outside of OctoPrint.

How can I do that?

jneilliii, 2023-11-25:
the only information I have on the library is from their build documentation, which is lack luster for linux environments.

https://github.com/prusa3d/libbgcode/blob/main/doc/building.md

fly74, 2023-11-25:
I live without it for the next time.

closed 2023-11-25 by fly74