jneilliii/OctoPrint-BGCode #6: Layer indicator not found

bradleyallanpatterson, 2023-11-25:
Just moved to latest prusa mini FW version 5.1.0.  I installed the latest BGODE plugin to OP and then sent an BGODE to OP.

Layer indicator not found in file: '02a-z-axis-motor-mount-side-p2_0.4n_0.2mm_PLA_MINI_48m.aw.bgcode'
Check 'layer pattern', 'Look ahead limit' in DisplayLayerProgress-Settings and reUpload the file!

Looking online, people say to add the below line to the custom Gode section but I still get the above error.

;LAYER:[layer_num]

jneilliii, 2023-11-26:
I'm sorry but I can't control/fix anything related to processing of the files once it's been converted from binary format to ascii. If you are getting unexpected results, I suggest you open issues with PrusaSlicer and any perspective plugin that is unable to process the file once it has been converted. This plugin is doing it's job with the tools provided by Prusa at this time.

bradleyallanpatterson, 2023-11-26:
How do I know if its PS or the plugin?

bradleyallanpatterson, 2023-11-28:
I disabled the Display Layer indicator plugin and OP still not working.   The exported bgcode prints fine from usb,..

jneilliii, 2023-11-29:
I'll need full logs then and example bgcode file to test.

jneilliii, 2023-11-29:
And prior to grabbing the logs and looking, enable the debug option for `octoprint.plugins.bgcode` in the logging settings of OctoPrint at the bottom.

[image]