jneilliii/OctoPrint-BGCode #7: Update to 0.2.0 rebuilds the Prusa library?

b-morgan, 2023-11-27:
Since I still had 0.1.0 installed, I got a notification that 0.2.0 was available so I updated it. Took quite a while to complete so I assume that it rebuilt the Prusa library. Is my assumption correct and did it need to do that?

jneilliii, 2023-11-27:
Yes, that assumption is correct until the Prusa3D team gets their pybgcode module registered with pypi for compiled distribution. It's an unfortunate caveat of how the module is being delivered by them currently.

jneilliii, 2023-11-27:
Hence my reply to the tags and release issue on their repo: https://github.com/prusa3d/libbgcode/issues/31#issuecomment-1826868724

closed 2023-11-27 by jneilliii

b-morgan, 2023-11-27:
As always you are on top of it! 😍