jneilliii/OctoPrint-BGCode #8: Unable to install on Octo4a

kdobrev, 2023-12-08:
Installing plugin "BGCode" from https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpog6oyfjg/OctoPrint-BGCode-master.zip --no-cache-dir
Processing /tmp/tmpog6oyfjg/OctoPrint-BGCode-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Collecting pybgcode@ https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
Downloading https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
| 474.1 kB 887.8 kB/s 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Installing backend dependencies: started
Installing backend dependencies: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'error'
error: subprocess-exited-with-error

× Preparing metadata (pyproject.toml) did not run successfully.
│ exit code: 1
╰─> [69 lines of output]
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building the dependencies with preset python-module
-- build dir = /tmp/pip-install-zvuvajdu/pybgcode_e6d2ddabf9ed4ae1a10430cff22c5e51/deps/build-python-module
CMake Error at /tmp/pip-install-zvuvajdu/pybgcode_e6d2ddabf9ed4ae1a10430cff22c5e51/deps/autobuild.cmake:62 (message):
Dependency configure failed with output:

CMake Error at
/tmp/pip-build-env-lvybkcut/normal/lib/python3.10/site-packages/cmake/data/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230
(message):

Could NOT find Git (missing: GIT_EXECUTABLE)

Call Stack (most recent call first):

/tmp/pip-build-env-lvybkcut/normal/lib/python3.10/site-packages/cmake/data/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
/tmp/pip-build-env-lvybkcut/normal/lib/python3.10/site-packages/cmake/data/share/cmake-3.27/Modules/FindGit.cmake:128 (find_package_handle_standard_args)
CMakeLists.txt:50 (find_package)

Call Stack (most recent call first):
/tmp/pip-install-zvuvajdu/pybgcode_e6d2ddabf9ed4ae1a10430cff22c5e51/CMakeLists.txt:27 (include)

-- Configuring incomplete, errors occurred!
Traceback (most recent call last):
File "/usr/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 144, in prepare_metadata_for_build_wheel
hook = backend.prepare_metadata_for_build_wheel
AttributeError: module 'py_build_cmake.build' has no attribute 'prepare_metadata_for_build_wheel'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/usr/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 351, in <module>
main()
File "/usr/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 333, in main
json_out['return_val'] = hook(**hook_input['kwargs'])
File "/usr/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 148, in prepare_metadata_for_build_wheel
whl_basename = backend.build_wheel(metadata_directory, config_settings)
File "/tmp/pip-build-env-lvybkcut/overlay/lib/python3.10/site-packages/py_build_cmake/build.py", line 67, in build_wheel
whl_name = self.build_wheel_in_dir(wheel_directory, tmp_build_dir,
File "/tmp/pip-build-env-lvybkcut/overlay/lib/python3.10/site-packages/py_build_cmake/build.py", line 248, in build_wheel_in_dir
self.do_native_cross_cmake_build(paths, cfg, pkg_info)
File "/tmp/pip-build-env-lvybkcut/overlay/lib/python3.10/site-packages/py_build_cmake/build.py", line 333, in do_native_cross_cmake_build
self.run_cmake(paths.source_dir, paths.staging_dir, cmake_cfg,
File "/tmp/pip-build-env-lvybkcut/overlay/lib/python3.10/site-packages/py_build_cmake/build.py", line 510, in run_cmake
cmaker.configure()
File "/tmp/pip-build-env-lvybkcut/overlay/lib/python3.10/site-packages/py_build_cmake/cmake.py", line 154, in configure
self.run(cmd, cwd=cwd, check=True, env=env)
File "/tmp/pip-build-env-lvybkcut/overlay/lib/python3.10/site-packages/py_build_cmake/cmake.py", line 68, in run
return self.runner.run(*args, **kwargs)
File "/tmp/pip-build-env-lvybkcut/overlay/lib/python3.10/site-packages/py_build_cmake/cmd_runner.py", line 25, in run
return sp_run(*args, **kwargs)
File "/usr/lib/python3.10/subprocess.py", line 526, in run
raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['cmake', '-S', '/tmp/pip-install-zvuvajdu/pybgcode_e6d2ddabf9ed4ae1a10430cff22c5e51/pybgcode', '-B', '/tmp/pip-install-zvuvajdu/pybgcode_e6d2ddabf9ed4ae1a10430cff22c5e51/.py-build-cmake_cache/cp310-cp310-linux_aarch64', '-D', 'PY_BUILD_CMAKE_PACKAGE_VERSION:STRING=0.2.0', '-D', 'PY_BUILD_CMAKE_PACKAGE_NAME:STRING=pybgcode', '-D', 'PY_BUILD_CMAKE_MODULE_NAME:STRING=pybgcode', '-D', 'Python3_EXECUTABLE:FILEPATH=/usr/bin/python3', '-D', 'Python3_ROOT_DIR:PATH=/usr', '-D', 'Python3_FIND_REGISTRY=NEVER', '-D', 'Python3_FIND_STRATEGY=LOCATION', '-D', 'PyBGCode_LINK_SYSTEM_LIBBGCODE=off', '-D', 'CMAKE_BUILD_TYPE:STRING=Release']' returned non-zero exit status 1.
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

jneilliii, 2023-12-08:
I don't control what this can and can't build on. I removed you just disable the bgcode option in the slicer.

kdobrev, 2023-12-10:
Hi, let's try to fix it together.

First error was: Could NOT find Git (missing: GIT_EXECUTABLE)
I solved it by logging in to the octo4a instance (or its linux sub machine) and doing:
`apk add git`

Second error is Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR) (Required is
at least version "1.0")

For this I did 
`apk add zlib`

did not work so I added

`apk add zlib-static
apk add zlib-ng`

but still nothing.

The library is obviously there, but it just cannot find it I think.

I tried these fixes: https://askubuntu.com/questions/1244299/cmake-can-not-find-zlib-on-ubuntu-20-04

but couldn't make it work.  Do you think you can have a try?

ncopa, 2023-12-11:
i think you may need install the `zlib-dev` package, which has the headers.

kdobrev, 2023-12-12:
yes, that fixed it.

Can you make the undelying OS install git and zlib-dev before installing the plugin, or should I contact the Octo4a developer with such suggestion?

jneilliii, 2023-12-13:
it's not really possible to enable OS system dependencies through the plugin infrastructure during install, but I'd be happy to add documentation to the readme about this. Just so I'm clear on the steps.

1. log in to the octo4a instance (or its linux sub machine) , not sure how this works, any pointers would be helpful.
2. apk add git
3. apk add zlib-dev

jneilliii, 2023-12-13:
I think I may have found the steps from octo4a site.

> After setting up OpenSSH server on settings tab, You can go to http://<your-ip>:5002 and use root username and the password you selected on the settings tab.

kdobrev, 2023-12-13:
Yes, this is it.

After setting up OpenSSH server on settings tab, You can go to http://:5002 and use root username and the password you selected on the settings tab.