jneilliii/OctoPrint-BGCode #11: Octo4a needs apk add cmake also

kdobrev, 2024-01-12:
Hi, 
I just installed a new octoprint on a phone and nhe first plug in I tried to install was bgcode, but it failed.

I checked the log and it appears I have previosly also added cmake.

Please add to the description of the plugin that 

apk add cmake

is also needed.
Thank you.

jneilliii, 2024-01-12:
How/where do you run that command? I've never used octo4a.

ruedli, 2024-03-02:
I think I run into this as well.

I have migrated from running on Pi (using octopi) to on proxmox in a VM.
I am _very_ happy on the performance of this compaired to Pi4, but:

All my other plugins installed well (more than 40) , but BGcode fails to compile.

- I tried running it in a container, with an octoprint install script provided from https://tteck.github.io/Proxmox/
- I tried on a debian VM, installing from scratch
- I tried with a VM script based on dietpi: see https://dietpi.com/docs/software/printing/#octoprint

All 3 environments worked OK, but failed to install this "one" plugin (i.e. BGCODE)

I first mention the output on a clean scripted container (between >>>>>>>> and <<<<<<)
And next after installing cmake. The last goed "slightly" better, but still errors out.

So: in a clean buld environemnt (restored from a Pi backup) I get this when adding the plugin: (message continues after the <<<<<<<<<<<< line)
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Installing plugin "BGCode" from https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip...
/opt/octoprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpps3fm8gu/OctoPrint-BGCode-master.zip --no-cache-dir
Processing /tmp/tmpps3fm8gu/OctoPrint-BGCode-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Collecting pybgcode@ https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip (from BGCode==0.2.0)
Downloading https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
- 474.1 kB 7.4 MB/s 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Installing backend dependencies: started
Installing backend dependencies: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'error'
error: subprocess-exited-with-error

× Preparing metadata (pyproject.toml) did not run successfully.
│ exit code: 1
╰─> [93 lines of output]
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building the dependencies with preset python-module
-- build dir = /tmp/pip-install-oodqq8vv/pybgcode_9b77c4829a0449d08c85e2daab589f42/deps/build-python-module
CMake Error at /tmp/pip-install-oodqq8vv/pybgcode_9b77c4829a0449d08c85e2daab589f42/deps/autobuild.cmake:72 (message):
Dependency build failed with output:

cmake -E tar: warning: unknown warning

cmake -E tar: warning: unknown warning

cmake -E tar: warning: unknown warning

cmake -E tar: warning: unknown warning

cmake -E tar: warning: unknown warning

cmake -E tar: warning: unknown warning

cmake -E tar: warning: unknown warning

[... 251 lines cut ...]
self.run_cmake(paths.source_dir, paths.staging_dir, cmake_cfg,
File "/tmp/pip-build-env-0vdphpb6/overlay/lib/python3.9/site-packages/py_build_cmake/build.py", line 510, in run_cmake
cmaker.configure()
File "/tmp/pip-build-env-0vdphpb6/overlay/lib/python3.9/site-packages/py_build_cmake/cmake.py", line 154, in configure
self.run(cmd, cwd=cwd, check=True, env=env)
File "/tmp/pip-build-env-0vdphpb6/overlay/lib/python3.9/site-packages/py_build_cmake/cmake.py", line 68, in run
return self.runner.run(*args, **kwargs)
File "/tmp/pip-build-env-0vdphpb6/overlay/lib/python3.9/site-packages/py_build_cmake/cmd_runner.py", line 25, in run
return sp_run(*args, **kwargs)
File "/usr/lib/python3.9/subprocess.py", line 528, in run
raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['cmake', '-S', '/tmp/pip-install-e97uf5jd/pybgcode_b1e56e28e261401585001da3f2c34aeb/pybgcode', '-B', '/tmp/pip-install-e97uf5jd/pybgcode_b1e56e28e261401585001da3f2c34aeb/.py-build-cmake_cache/cp39-cp39-linux_x86_64', '-D', 'PY_BUILD_CMAKE_PACKAGE_VERSION:STRING=0.2.0', '-D', 'PY_BUILD_CMAKE_PACKAGE_NAME:STRING=pybgcode', '-D', 'PY_BUILD_CMAKE_MODULE_NAME:STRING=pybgcode', '-D', 'Python3_EXECUTABLE:FILEPATH=/opt/octoprint/bin/python3', '-D', 'Python3_ROOT_DIR:PATH=/opt/octoprint', '-D', 'Python3_FIND_REGISTRY=NEVER', '-D', 'Python3_FIND_STRATEGY=LOCATION', '-D', 'PyBGCode_LINK_SYSTEM_LIBBGCODE=off', '-D', 'CMAKE_BUILD_TYPE:STRING=Release']' returned non-zero exit status 1.
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

jneilliii, 2024-03-02:
well, apparently this guy has patched the cmake for octo4a compatibility.

https://github.com/kdobrev/libbgcode/tree/master

In either case, I'd recommend you guys look at the building instructions for potential missing dependencies/requirements for your OS.

https://github.com/prusa3d/libbgcode/blob/main/doc/building.md

jneilliii, 2024-03-02:
I've merged those changes into my fork of libbgcode and the latest updates from the upstream repo and am testing on pi4 now to make sure it's still working.

jneilliii, 2024-03-02:
still works on pi 4 as expected.

jneilliii, 2024-03-02:
updated octo4a instructions in readme and about to merge the changes to the plugin repository. @ruedli I think you issue is different.

closed 2024-03-02 by jneilliii

ruedli, 2024-03-14:
Just for anyone that runs into this when installing it on ubuntu on proxmox: I figured out what I was missing.

I started with mentioned Proxmox VM from Dietpi: I tried with a VM script based on dietpi: see https://dietpi.com/docs/software/printing/#octoprint

You can select octoprint directly from the dietpi-software and recover from an octoprint backup.

While this configured all my plugins for octoprint, not BGCODE.

In oreder to install I had to (from root):
apt install git
apt install zlib1g
apt install zlib1g-dev
apt install build-essential

I also installed (but these seemed to be installed already):
apt install cmake
apt install gcc

Further I adapted .bashrc in both /root and /home/dietpi and included a line:
export CXX=/usr/bin/gcc

The dietpi documentation suggests that you could instead configure /etc/bashrc.d/dietpi-octoprint.sh but I did not verify this.

The vm on my n100 processor is incredibly fast compared to running on a Pi4. Besides that it can run other softwere: my homeassistant (with mqtt server and node-red from which you can interact with the printer) and pihole with (recursive dns), run in containers and there is space for additional windows 11 and ubuntu vm's.

Impressive