jneilliii/OctoPrint-BGCode #12: BGCode install

IanAdd, 2024-03-09:
Running octoprint on a Pi.
It has worked well for a few years talking to a Prusa Mk3S+.
I've just upgraded the Prusa to Mk3.5 where Prusaslicer delivers BGCode. 
So, I select your BGCode plugin to install. It says successfully installed at end but does not show up in the plugin list.

I reran the install, same problem, here is the log, 
I am running octoprint 1.9.3

Any suggestions?

```
Installing plugin "BGCode" from https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpouu0au6b/OctoPrint-BGCode-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpouu0au6b/OctoPrint-BGCode-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from BGCode==0.2.0) (1.9.3)
Collecting pybgcode@ https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
Downloading https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Installing backend dependencies: started
Installing backend dependencies: finished with status 'done'
Preparing wheel metadata: started
Preparing wheel metadata: still running...
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python3 /home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpa_ejs__u
cwd: /tmp/pip-install-c53g833p/pybgcode_a9d16bc4b15d4a11b6a4ce2b01f2cde9
Complete output (74 lines):
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building the dependencies with preset python-module
-- build dir = /tmp/pip-install-c53g833p/pybgcode_a9d16bc4b15d4a11b6a4ce2b01f2cde9/deps/build-python-module
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found suitable version "1.2.11", minimum required is "1.0")
-- Found Boost: /tmp/pip-install-c53g833p/pybgcode_a9d16bc4b15d4a11b6a4ce2b01f2cde9/deps/build-python-module/destdir/usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found suitable version "1.82.0", minimum required is "1.78")
CMake Warning (dev) at /tmp/pip-build-env-7tbc5j1n/normal/lib/python3.7/site-packages/cmake/data/share/cmake-3.28/Modules/FindZLIB.cmake:88 (message):
ZLIB does not provide any COMPONENTS.  Calling

find_package(ZLIB COMPONENTS ...)

will always fail.
Call Stack (most recent call first):
/tmp/pip-install-c53g833p/pybgcode_a9d16bc4b15d4a11b6a4ce2b01f2cde9/CMakeLists.txt:120 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /tmp/pip-build-env-7tbc5j1n/normal/lib/python3.7/site-packages/cmake/data/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
Could NOT find ZLIB (missing: 1.0) (found version "1.2.11")
Call Stack (most recent call first):
/tmp/pip-build-env-7tbc5j1n/normal/lib/python3.7/site-packages/cmake/data/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
/tmp/pip-build-env-7tbc5j1n/normal/lib/python3.7/site-packages/cmake/data/share/cmake-3.28/Modules/FindZLIB.cmake:199 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
/tmp/pip-install-c53g833p/pybgcode_a9d16bc4b15d4a11b6a4ce2b01f2cde9/CMakeLists.txt:120 (find_package)

-- Configuring incomplete, errors occurred!
Traceback (most recent call last):
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 126, in prepare_metadata_for_build_wheel
hook = backend.prepare_metadata_for_build_wheel
AttributeError: module 'py_build_cmake.build' has no attribute 'prepare_metadata_for_build_wheel'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
main()
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
json_out['return_val'] = hook(**hook_input['kwargs'])
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 131, in prepare_metadata_for_build_wheel
config_settings)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 159, in _get_wheel_metadata_from_wheel
whl_basename = backend.build_wheel(metadata_directory, config_settings)
File "/tmp/pip-build-env-7tbc5j1n/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 68, in build_wheel
config_settings)
File "/tmp/pip-build-env-7tbc5j1n/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 248, in build_wheel_in_dir
self.do_native_cross_cmake_build(paths, cfg, pkg_info)
File "/tmp/pip-build-env-7tbc5j1n/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 334, in do_native_cross_cmake_build
cfg.cross, package_info, native_install_dir)
File "/tmp/pip-build-env-7tbc5j1n/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 510, in run_cmake
cmaker.configure()
File "/tmp/pip-build-env-7tbc5j1n/overlay/lib/python3.7/site-packages/py_build_cmake/cmake.py", line 154, in configure
self.run(cmd, cwd=cwd, check=True, env=env)
File "/tmp/pip-build-env-7tbc5j1n/overlay/lib/python3.7/site-packages/py_build_cmake/cmake.py", line 68, in run
return self.runner.run(*args, **kwargs)
File "/tmp/pip-build-env-7tbc5j1n/overlay/lib/python3.7/site-packages/py_build_cmake/cmd_runner.py", line 25, in run
return sp_run(*args, **kwargs)
File "/usr/lib/python3.7/subprocess.py", line 487, in run
output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['cmake', '-S', '/tmp/pip-install-c53g833p/pybgcode_a9d16bc4b15d4a11b6a4ce2b01f2cde9/pybgcode', '-B', '/tmp/pip-install-c53g833p/pybgcode_a9d16bc4b15d4a11b6a4ce2b01f2cde9/.py-build-cmake_cache/cp37-cp37m-linux_armv7l', '-D', 'PY_BUILD_CMAKE_PACKAGE_VERSION:STRING=0.2.0', '-D', 'PY_BUILD_CMAKE_PACKAGE_NAME:STRING=pybgcode', '-D', 'PY_BUILD_CMAKE_MODULE_NAME:STRING=pybgcode', '-D', 'Python3_EXECUTABLE:FILEPATH=/home/pi/oprint/bin/python3', '-D', 'Python3_ROOT_DIR:PATH=/home/pi/oprint', '-D', 'Python3_FIND_REGISTRY=NEVER', '-D', 'Python3_FIND_STRATEGY=LOCATION', '-D', 'PyBGCode_LINK_SYSTEM_LIBBGCODE=off', '-D', 'CMAKE_BUILD_TYPE:STRING=Release']' returned non-zero exit status 1.
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python3 /home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpa_ejs__u Check the logs for full command output.
Preparing wheel metadata: finished with status 'error'
Looks like the plugin was already installed. Forcing a reinstall.
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpouu0au6b/OctoPrint-BGCode-master.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpouu0au6b/OctoPrint-BGCode-master.zip
Building wheels for collected packages: BGCode
Building wheel for BGCode (setup.py): started
Building wheel for BGCode (setup.py): finished with status 'done'
Created wheel for BGCode: filename=BGCode-0.2.0-py2.py3-none-any.whl size=4225 sha256=264e336041dc5a3677914db6bd7acac7baa5c02f356904057226894097a3658e
Stored in directory: /tmp/pip-ephem-wheel-cache-655xtwn_/wheels/59/80/2b/2f1f7301db1b93a7512beda8b655c851cdb369cee3a694982f
Successfully built BGCode
Installing collected packages: BGCode
Successfully installed BGCode-0.2.0
Done!

jneilliii, 2024-03-10:
try installing zlib with the following command in SSH and then reinstall the plugin. 
```
sudo apt install zlib-dev
```

jneilliii, 2024-03-10:
The other option is just to disable the bgcode format from the slicer completely and not use that format. you don't gain anything from it with OctoPrint. 

[image]

skardach, 2024-03-10:
I believe the issue might be here: https://github.com/jneilliii/libbgcode/blob/main/CMakeLists.txt#L120

[Apparently](https://github.com/microsoft/vcpkg/issues/31994#issuecomment-1592392216) version argument has to be on the second argument to `find_package` as shown here: https://cmake.org/cmake/help/latest/command/find_package.html#basic-signature. In the issue I linked they mention CMake 3.26 as the version that started enforcing this.

Interestingly enough on Alpine linux chroot used for octo4a if one does not install `cmake`, this issue will pop-up (perhaps python's wheel installs cmake into a temporary location?). Once I installed Cmake (packaged version is 3.24.4), this error went away.

IanAdd, 2024-03-12:
> try installing zlib with the following command in SSH and then reinstall the plugin.
> 
> ```
> sudo apt install zlib-dev
> ```
E: Unable to locate package zlib-dev

IanAdd, 2024-03-12:
> The other option is just to disable the bgcode format from the slicer completely and not use that format. you don't gain anything from it with OctoPrint.
> 
When I revert prusaslicer to gcode from bgcode I see that prusalink, should I decide to ask it to print instead of octoprint, does not draw the picture of the object being printed, so I'd like to use bgcode for that reason alone at this stage.

IanAdd, 2024-03-13:
> Embedded thumbnails in gcode doesn't work with the new firmware? My Slicer Thumbnails plugin can extract those from the gcode file on upload to OctoPrint. If your slicer is configured for OctoPrint instead of PrusaLink you should be able to send directly to OctoPrint, but I don't understand the first part of your statement.
> 
> > When I revert prusaslicer to gcode from bgcode I see that prusalink

If I drop a gcode file into prusalink it does not show the image, if i drop a bgcode one in it does show the image. See screenshot.

[image]

jneilliii, 2024-03-15:
I wonder if these instructions might help you....extracted from https://github.com/jneilliii/OctoPrint-BGCode/issues/11#issuecomment-1997589971

```
sudo apt install git
sudo apt install zlib1g
sudo apt install zlib1g-dev
sudo apt install build-essential
```

IanAdd, 2024-03-16:
> I wonder if these instructions might help you....extracted from [#11 (comment)](https://github.com/jneilliii/OctoPrint-BGCode/issues/11#issuecomment-1997589971)
> 
> ```
> sudo apt install git
> sudo apt install zlib1g
> sudo apt install zlib1g-dev
> sudo apt install build-essential
> ```

All are already installed

Flatstrap, 2024-04-23:
Hi, I had the same issue, upgrading from an old install of octoprint when upgrading from a Prusa MK3s+ to a Mk3.9.
To resolve I SSH'd in and ran these commands (I hadn't updated for 18months!):
sudo apt update
sudo apt upgrade -y
then restarted the raspberrypi and re-installed BGCODE from within octoprint and it worked fine, although it does take a bout 15 minutes so be patient.

IanAdd, 2024-04-25:
@Flatstrap , thanks for the tip but it didn't help.
Did update & upgrade, rebooted
Attempt install BGcode from UI and slightly different error list due to OS updates, I guess ....

Installing plugin "BGCode" from https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpdg8hd_vp/OctoPrint-BGCode-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpdg8hd_vp/OctoPrint-BGCode-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from BGCode==0.2.0) (1.10.0)
Collecting pybgcode@ https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
Downloading https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Preparing wheel metadata: started
Preparing wheel metadata: still running...
Preparing wheel metadata: still running...
Preparing wheel metadata: still running...
Preparing wheel metadata: finished with status 'error'
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python3 /home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp8yj_txsp
cwd: /tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1
Complete output (100 lines):
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building the dependencies with preset python-module
-- build dir = /tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1/deps/build-python-module
CMake Warning at ../deps/autobuild.cmake:48 (message):
CMake presets are not supported with this version of CMake.  Building all
dependency packages!
Call Stack (most recent call first):
../CMakeLists.txt:27 (include)

-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Found ZLIB: /tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1/deps/build-python-module/destdir/usr/local/lib/libz.a (found suitable version "1.2.11", minimum required is "1.0")
-- Found Boost: /tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1/deps/build-python-module/destdir/usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found suitable version "1.82.0", minimum required is "1.78")
-- Found ZLIB: /tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1/deps/build-python-module/destdir/usr/local/lib/libz.a (found version "1.2.11")
-- Found Python3: /home/pi/oprint/bin/python3 (found version "3.7.3") found components: Interpreter Development
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1/deps/build-python-module/destdir/usr/local/include (found version "2.11.1")
-- Found Boost: /tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1/deps/build-python-module/destdir/usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0") found components: nowide
-- Configuring done
-- Generating done
CMake Warning:
Manually-specified variables were not used by the project:

PY_BUILD_CMAKE_PACKAGE_NAME

-- Build files have been written to: /tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1/.py-build-cmake_cache/cp37-cp37m-linux_armv7l
Scanning dependencies of target bgcode_core
[ 11%] Building CXX object upstream/src/LibBGCode/core/CMakeFiles/bgcode_core.dir/core.cpp.o
[... 42 lines cut ...]
cmaker.build()
File "/tmp/pip-build-env-r2wzv11m/overlay/lib/python3.7/site-packages/py_build_cmake/cmake.py", line 190, in build
self.run(cmd, cwd=cwd, check=True, env=env)
File "/tmp/pip-build-env-r2wzv11m/overlay/lib/python3.7/site-packages/py_build_cmake/cmake.py", line 68, in run
return self.runner.run(*args, **kwargs)
File "/tmp/pip-build-env-r2wzv11m/overlay/lib/python3.7/site-packages/py_build_cmake/cmd_runner.py", line 25, in run
return sp_run(*args, **kwargs)
File "/usr/lib/python3.7/subprocess.py", line 487, in run
output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['cmake', '--build', '/tmp/pip-install-v3288la_/pybgcode_69c4b944712c4a618a122f37089b0cb1/.py-build-cmake_cache/cp37-cp37m-linux_armv7l', '--config', 'Release', '--target', 'build_and_test']' returned non-zero exit status 2.
----------------------------------------
[... cut ...]
Building wheels for collected packages: BGCode
Building wheel for BGCode (setup.py): started
Building wheel for BGCode (setup.py): finished with status 'done'
Created wheel for BGCode: filename=BGCode-0.2.0-py2.py3-none-any.whl size=4225 sha256=2c5e53ea9718ae9d1b486420de92bf9ef65ebf4d687f75e71396aa20ab54e9c8
Stored in directory: /tmp/pip-ephem-wheel-cache-5l0gp0bj/wheels/6d/dc/f7/ecb04e9f4a2573bf902ac057f9fa4ea25053e986ad21f9610e
Successfully built BGCode
Installing collected packages: BGCode
Successfully installed BGCode-0.2.0
Done!

elpeter25, 2024-05-02:
Also having issues installing it.
Octoprint running in Ubuntu with several instances of octoprint running on it. See error below:

`Installing plugin "BGCode" from https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip...
/home/octoprint/python_venvs/octoprint3/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpvjtqxxy7/OctoPrint-BGCode-master.zip --no-cache-dir
Processing /tmp/tmpvjtqxxy7/OctoPrint-BGCode-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Collecting pybgcode@ https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip (from BGCode==0.2.0)
Downloading https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
\ 475.9 kB 3.5 MB/s 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Installing backend dependencies: started
Installing backend dependencies: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'error'
error: subprocess-exited-with-error

× Preparing metadata (pyproject.toml) did not run successfully.
│ exit code: 1
╰─> [65 lines of output]
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building the dependencies with preset python-module
-- build dir = /tmp/pip-install-k45d1_o2/pybgcode_cc4b86e483604e77a8710b65171d9fd7/deps/build-python-module
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found suitable version "1.2.11", minimum required is "1.0")
-- Found Boost: /tmp/pip-install-k45d1_o2/pybgcode_cc4b86e483604e77a8710b65171d9fd7/deps/build-python-module/destdir/usr/local/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found suitable version "1.82.0", minimum required is "1.78")
CMake Warning (dev) at /tmp/pip-build-env-euxcxv74/normal/lib/python3.7/site-packages/cmake/data/share/cmake-3.29/Modules/FindZLIB.cmake:88 (message):
ZLIB does not provide any COMPONENTS.  Calling

find_package(ZLIB COMPONENTS ...)

will always fail.
Call Stack (most recent call first):
/tmp/pip-install-k45d1_o2/pybgcode_cc4b86e483604e77a8710b65171d9fd7/CMakeLists.txt:120 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /tmp/pip-build-env-euxcxv74/normal/lib/python3.7/site-packages/cmake/data/share/cmake-3.29/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
Could NOT find ZLIB (missing: 1.0) (found version "1.2.11")
Call Stack (most recent call first):
/tmp/pip-build-env-euxcxv74/normal/lib/python3.7/site-packages/cmake/data/share/cmake-3.29/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
/tmp/pip-build-env-euxcxv74/normal/lib/python3.7/site-packages/cmake/data/share/cmake-3.29/Modules/FindZLIB.cmake:199 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
/tmp/pip-install-k45d1_o2/pybgcode_cc4b86e483604e77a8710b65171d9fd7/CMakeLists.txt:120 (find_package)

-- Configuring incomplete, errors occurred!
Traceback (most recent call last):
File "/home/octoprint/python_venvs/octoprint3/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
main()
File "/home/octoprint/python_venvs/octoprint3/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
json_out['return_val'] = hook(**hook_input['kwargs'])
File "/home/octoprint/python_venvs/octoprint3/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
[... 5 lines cut ...]
File "/tmp/pip-build-env-euxcxv74/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 334, in do_native_cross_cmake_build
cfg.cross, package_info, native_install_dir)
File "/tmp/pip-build-env-euxcxv74/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 510, in run_cmake
cmaker.configure()
File "/tmp/pip-build-env-euxcxv74/overlay/lib/python3.7/site-packages/py_build_cmake/cmake.py", line 154, in configure
self.run(cmd, cwd=cwd, check=True, env=env)
File "/tmp/pip-build-env-euxcxv74/overlay/lib/python3.7/site-packages/py_build_cmake/cmake.py", line 68, in run
return self.runner.run(*args, **kwargs)
File "/tmp/pip-build-env-euxcxv74/overlay/lib/python3.7/site-packages/py_build_cmake/cmd_runner.py", line 25, in run
return sp_run(*args, **kwargs)
File "/usr/lib/python3.7/subprocess.py", line 512, in run
output=stdout, stderr=stderr)
[... cut ...] BGCODE=off', '-D', 'CMAKE_BUILD_TYPE:STRING=Release']' returned non-zero exit status 1.
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`

jneilliii, 2024-05-03:
I didn't make the underlying package, PrusaResearch did. Whatever build dependencies that are necessary for building it from their repo should give you a hint as to what you might need to install the plugin successfully. The one that seems to stand out is missing zlib-dev. So you could try `sudo apt install zlib-dev` and see if that helps, otherwise see how to build the underlying library and all it's dependencies from [here](https://github.com/prusa3d/libbgcode).

MichaIng, 2024-05-18:
On Debian, Ubuntu etc, it's `zlib1g-dev` 🙂.

Would you be open to upload the plugin to PyPI, in case even with pre-compiled wheels? I could help with GitHub Actions workflows to do this either triggered, or automatically when a new release is tagged.

jneilliii, 2024-05-18:
I'm totally open to that, but really wish the PrusaResearch team would do it on their repo so I can just have it as a dependency in my plugin. For the actions I assume those would be made against the other repo I've forked from them. I was using a fork since there's no telling what they are changing without any release/versioning happening. 

[https://github.com/jneilliii/libbgcode/](https://github.com/jneilliii/libbgcode/)

jneilliii, 2024-05-18:
ideally, compatible with 32bit octopi images...

MichaIng, 2024-05-19:
> I'm totally open to that, but really wish the PrusaResearch team would do it on their repo so I can just have it as a dependency in my plugin.

Oh that's true, I did not recognise this dependency. Hmm, of course you could theoretically publish libbgcode as fork on PyPI as well, but that is not optimal. I found the requests upstream:
- https://github.com/prusa3d/libbgcode/issues/44
- https://github.com/prusa3d/libbgcode/issues/31

They even have the build workflows already: https://github.com/prusa3d/libbgcode/actions/runs/8466159186
Missing are ARM builds and upload via to PyPI via `twine`. Cross-compiling/emulation for ARM builds can be easily done with https://github.com/pypa/cibuildwheel. The only thing I am not 100% sure about is which glibc versions are used in `cibuildwheel` and in case how to control this, to make the wheels compatible with all relevant distributions, or have multiple wheels to cover all of them.

We could edit the workflow on your fork to test all this and do uploads to https://test.pypi.org/ with a different module name (to avoid conflicts with PrusaResearch, in case they want to do the same). And when everything works well, do a PR. Since there is no reaction on the two linked issues, a PR seems to be the only chance to bring things forward.

jneilliii, 2024-05-20:
> We could edit the workflow on your fork to test all this and do uploads to [test.pypi.org](https://test.pypi.org/) with a different module name (to avoid conflicts with PrusaResearch, in case they want to do the same). And when everything works well, do a PR. Since there is no reaction on the two linked issues, a PR seems to be the only chance to bring things forward.

I'm cool with that approach. I'll add you as a contributor to my fork to make this process a little easier.

cavernerg, 2024-09-11:
sudo apt install cmake 
fixed it for me (on a fully updated ubuntu)

elpeter25, 2025-01-08:
@MichaIng @jneilliii any progress on this?

Thanks so much in advance!

MichaIng, 2025-01-08:
Coincidentally I found some time today to review this issue, and have the GitHub Actions workflow open in my notepad++ right now 😄.

But I did not yet start to implement the changes. First, `cmake --build` needs to run, I guess this option help to do that within the cross-arch builds, before each wheel: https://cibuildwheel.pypa.io/en/stable/options/#before-build

Will most likely open a PR tomorrow.

M1Sports20, 2025-01-24:
I don't really know what you have to pass to `cmake --build`.  I will submit a PR, if I can figure it out.

jneilliii, 2025-01-24:
@MichaIng has already started on the PR here: https://github.com/jneilliii/libbgcode/pull/1

can't remember if it's ready or not to merge TBH.

MichaIng, 2025-01-24:
Ah, I wanted to split off the `Development.Module` and concurrency avoidance PRs, so we can offer it upstream in smaller chunks with more targeted aim. Will do tomorrow.

MichaIng, 2025-04-30:
"Tomorrow", well, better late than never 😅: https://github.com/prusa3d/libbgcode/pull/58

I am a bit worried about the low maintainer interaction. Enabling wheel builds with manylinux is the absolute basis for automated multi-arch/multi-platform/multi-Python wheel builds the way I can implement it into their workflows. Let's hope they consider it.

Otherwise we could upload own wheels under a different project name, and in any case to https://test.pypi.org to test the workflow after upload steps have been added. There were still some issues with certain platform/arch combinations using cibuildwheel, but the most important ones work well: https://github.com/jneilliii/libbgcode/actions/runs/14025192546

jneilliii, 2025-07-06:
I'd be down with that, just not familiar with pypi builds enough to know what is needed to change in pyproject.

MichaIng, 2025-07-07:
No change really needed, but to be fair and transparent, I suggest:
- Give the project `name` a suffix to not block the original project name slot on PyPI
- `Homepage` URL, so people see where it is hosted, and transparently visible as fork on GitHub
- A note at the top of the readme, that it is a soft fork with the aim to provide release tags and wheels on PyPI _and whichever other intention I may have missed_. And that we are offering the related commits and workflows to Prusa, if they want to distribute via PyPI themselves.

MichaIng, 2025-07-07:
I see you are playing with a testpypi job already. Just for comparison, here is how we push releases in motionEye via `twine`: https://github.com/motioneye-project/motioneye/blob/main/.github/workflows/pypi_release.yml

So what is needed is an account on testpypi (and pypi, not sure whether it is one account for both platforms), then create a token there, save it as Actions secret in repository settings.

I guess this step could just be added as step to the existing cibuildwheel job. I think multiple dedicated `twine` calls for each matrix call will add to the same release at PyPI, but to make this sure is why testpypi is great to use first 🙂.

MichaIng, 2025-07-07:
Right, the license classifier was deprecated, and also `license` does not take a file anymore. Use this instead:
```ini
license = "AGPL-3.0-or-later"
license-files = ["LICENSE"]
```
https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license-and-license-files

And regarding the build and release workflow:
- I think we want builds to always run on push and in case PR, to recognize failures early, and not when it is too late after the release has been pushed already.
- But of course we do not want PyPI uploads on every push. The `release` event is theoretically a solution, but it would mean concurrent builds on releases, once for the `push`, once for the `release`. And if for some reason we need to update the tag, it wouldn't run again.
- This condition should do for the upload step, using again `push` trigger instead of `release`:
  ```yaml
  if github.event_name == 'push' && github.ref_type == 'tag'
  ```
  So it runs only on push events to an actual tag.

jneilliii, 2025-07-07:
Thanks @MichaIng for the suggestions. I've implemented those changes, but the publish to pypi still seems unhappy about the license. 

https://github.com/jneilliii/libbgcode-jneilliii/actions/runs/16120080872/job/45484091329

jneilliii, 2025-07-08:
I just release version 0.2.2 that is now using pre-compiled builds from pypi. Hopefully installs will now be much smoother and very quick compared to having to build every update.