jneilliii/OctoPrint-BGCode #13: raspberry pi zero 2w

zubzero689, 2024-04-12:
now to start off i have this bgcode installed on 3b+ and a 4b i was wanting to free up my 4b with my zero 2w but the plugin wont install no matter how i try.  i tried in the UI and it just ends up crashing out and i have to restart the zero 2w i did leave it for over 24 hours when this happened the first time just to make sure it had the time.  when i try it thru SSH it errors out 
pi@pi-hole-2w:~ $ pip install "https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip
  Using cached https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip
    ERROR: Command errored out with exit status 255:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-ze354lxx/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-ze354lxx/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-kl_w6_y0
         cwd: /tmp/pip-req-build-ze354lxx/
    Complete output (1 lines):
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
    ----------------------------------------
WARNING: Discarding https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip. Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 255: python setup.py egg_info Check the logs for full command output.

now i have made this zero 2w fully up to date and i still get this error at this point i just think the zero 2w insnt compatable

zubzero689, 2024-04-12:
thanks for the info very helpful i did the above and told me the pip need updating so i did that and now i get stuck on 
(oprint) pi@pi-hole-2w:~ $ pip install https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip
  Downloading https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip
     \ 23.7 kB 851.8 kB/s 0:00:00
  Preparing metadata (setup.py) ... done
Collecting pybgcode@ https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip (from BGCode==0.2.0)
  Downloading https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
     / 475.9 kB 1.8 MB/s 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... /

now i was monitoring resources till it stopped outputting and the ram was almost used up but everything else was normal cpu wasnt at high usage network wasnt ether.
I understand if this plugin wont work on pi zero2w i just wanted people to be aware if they come here.
I also understand if you want to mark this as closed this seems to be the end of the line for now.
thanks for the help again

jneilliii, 2024-04-12:
The 2w might be running out of memory during the compile process. I think there's a config.txt line you can add to help with that. Will have to dig it up from one of my other plugins that had similar issues with numpy.

jneilliii, 2024-04-12:
Found it, https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/141#issuecomment-542227338. See if this works. After booting up the pi SSH in and run these commands.

Add additional swap memory
```
sudo fallocate -l 1G /swapfile
sudo chmod 600 /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
```

remove additional swap memory after installation and rebooting.
```
sudo swapoff -v /swapfile
sudo rm /swapfile
```

zubzero689, 2024-04-13:
thank you again it all is working right now as intended

closed 2024-04-13 by zubzero689