jneilliii/OctoPrint-BGCode #14: Can't install libssl.so.3 error

aharbick, 2024-06-10:
I've tried to install this plugin and it fails (I believe) because pip can't find libssl.so.3.  Any suggestions?

```
Installing plugin "BGCode" from https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpfi41ngx3/OctoPrint-BGCode-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpfi41ngx3/OctoPrint-BGCode-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from BGCode==0.2.0) (1.10.1)
Collecting pybgcode@ https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
Downloading https://github.com/jneilliii/libbgcode/archive/refs/tags/0.2.0.zip
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Installing backend dependencies: started
Installing backend dependencies: finished with status 'done'
Preparing wheel metadata: started
Preparing wheel metadata: finished with status 'error'
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python3 /home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp1eumbo3z
cwd: /tmp/pip-install-cswno0jg/pybgcode_cf7c868621324c1b90aaddf56217e741
Complete output (34 lines):
/tmp/pip-build-env-_nibmco4/normal/lib/python3.7/site-packages/cmake/data/bin/cmake: error while loading shared libraries: libssl.so.3: cannot open shared object file: No such file or directory
Traceback (most recent call last):
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 126, in prepare_metadata_for_build_wheel
hook = backend.prepare_metadata_for_build_wheel
AttributeError: module 'py_build_cmake.build' has no attribute 'prepare_metadata_for_build_wheel'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
main()
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
json_out['return_val'] = hook(**hook_input['kwargs'])
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 131, in prepare_metadata_for_build_wheel
config_settings)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 159, in _get_wheel_metadata_from_wheel
whl_basename = backend.build_wheel(metadata_directory, config_settings)
File "/tmp/pip-build-env-_nibmco4/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 68, in build_wheel
config_settings)
File "/tmp/pip-build-env-_nibmco4/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 248, in build_wheel_in_dir
self.do_native_cross_cmake_build(paths, cfg, pkg_info)
File "/tmp/pip-build-env-_nibmco4/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 334, in do_native_cross_cmake_build
cfg.cross, package_info, native_install_dir)
File "/tmp/pip-build-env-_nibmco4/overlay/lib/python3.7/site-packages/py_build_cmake/build.py", line 510, in run_cmake
cmaker.configure()
File "/tmp/pip-build-env-_nibmco4/overlay/lib/python3.7/site-packages/py_build_cmake/cmake.py", line 154, in configure
self.run(cmd, cwd=cwd, check=True, env=env)
File "/tmp/pip-build-env-_nibmco4/overlay/lib/python3.7/site-packages/py_build_cmake/cmake.py", line 68, in run
return self.runner.run(*args, **kwargs)
File "/tmp/pip-build-env-_nibmco4/overlay/lib/python3.7/site-packages/py_build_cmake/cmd_runner.py", line 25, in run
return sp_run(*args, **kwargs)
File "/usr/lib/python3.7/subprocess.py", line 487, in run
output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['cmake', '-S', '/tmp/pip-install-cswno0jg/pybgcode_cf7c868621324c1b90aaddf56217e741/pybgcode', '-B', '/tmp/pip-install-cswno0jg/pybgcode_cf7c868621324c1b90aaddf56217e741/.py-build-cmake_cache/cp37-cp37m-linux_armv7l', '-D', 'PY_BUILD_CMAKE_PACKAGE_VERSION:STRING=0.2.0', '-D', 'PY_BUILD_CMAKE_PACKAGE_NAME:STRING=pybgcode', '-D', 'PY_BUILD_CMAKE_MODULE_NAME:STRING=pybgcode', '-D', 'Python3_EXECUTABLE:FILEPATH=/home/pi/oprint/bin/python3', '-D', 'Python3_ROOT_DIR:PATH=/home/pi/oprint', '-D', 'Python3_FIND_REGISTRY=NEVER', '-D', 'Python3_FIND_STRATEGY=LOCATION', '-D', 'PyBGCode_LINK_SYSTEM_LIBBGCODE=off', '-D', 'CMAKE_BUILD_TYPE:STRING=Release']' returned non-zero exit status 127.
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python3 /home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp1eumbo3z Check the logs for full command output.
Looks like the plugin was already installed. Forcing a reinstall.
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpfi41ngx3/OctoPrint-BGCode-master.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpfi41ngx3/OctoPrint-BGCode-master.zip
Building wheels for collected packages: BGCode
Building wheel for BGCode (setup.py): started
Building wheel for BGCode (setup.py): finished with status 'done'
Created wheel for BGCode: filename=BGCode-0.2.0-py2.py3-none-any.whl size=4225 sha256=405d3c8e93b19e4ca2b5b734425989c4f71fe25d3a720774e984e9b9d051d18e
Stored in directory: /tmp/pip-ephem-wheel-cache-5y_b39zt/wheels/78/99/85/21aa7fd9a24221009eeb1fca69fd2529b60c11bb1d0bbbba1e
Successfully built BGCode
Installing collected packages: BGCode
Successfully installed BGCode-0.2.0
Done!

```

jneilliii, 2024-06-10:
try this in SSH session of your pi.

https://askubuntu.com/questions/339364/libssl-so-10-cannot-open-shared-object-file-no-such-file-or-directory

aharbick, 2024-06-10:
🤔 in not clear what you’re suggesting…. That I link a version 1.0 of the libssl lib to libssl.so.3?  The problem in the linked post was that some code was incorrectly linking against libssl.so.10 but that it was actually the same version/dependency.

I think my case is a version mismatch for some reason…

jneilliii, 2024-06-10:
I meant, just make sure it's installed. 

```
sudo apt-get update
sudo apt-get install libssl1.0.0 libssl-dev
```

aharbick, 2024-06-11:
Yeah.  I ran that and it is installed.  But it still can’t find libssl.so.3.  I forced symlinks to that (as well as libcrypto.so.3) and then it fails trying to find OpenSSL 3.0 symbols.  I think something has a dependency on a newer version of OpenSSL. 🤷‍♂️

jneilliii, 2024-06-11:
yeah, the compiling side of libgcode is beyond me and prusa doesn't really give any good insight from their repo [here](https://github.com/prusa3d/libbgcode) for that module/library.

aharbick, 2024-06-11:
Well thanks for looking anyhow!  I build openssl3 from scratch with this https://hostnextra.com/learn/tutorials/how-to-install-openssl-3-on-ubuntu

and then I get a different linker error with libstdc++6

I’ll keep digging and see if I can figure something out and report back.

badhex, 2024-06-21:
What fixed my octopi was ssh'ing into my pi and executing:

- sudo apt update
- sudo apt upgrade

aharbick, 2025-09-17:
Looking like maybe I need to update Octoprint.... AI said "The BGCode plugin now depends on a package called pybgcode-jneilliii that’s published on PyPI only as wheels for newer Python versions (3.10–3.14) and no source distribution—so pip on 3.7 can’t install it"

The UI is also giving a warning.  I'll try to upgrade.

jneilliii, 2025-09-17:
Plugin listing and install requirements of the plugin have been updated to limit to minimum python version of 3.8. Simplest update procedure would be to create backup in OctoPrint's settings, download backup, flash latest OctoPi image on new SD card, and use the backup during the initial setup wizard of OctoPrint to restore everything.

aharbick, 2025-09-17:
For the record this was the output

```
nstalling plugin "BGCode" from https://github.com/jneilliii/OctoPrint-BGCode/archive/master.zip...
/opt/octopi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmptjaj1ils/OctoPrint-BGCode-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmptjaj1ils/OctoPrint-BGCode-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /opt/octopi/oprint/lib/python3.11/site-packages (from BGCode==0.2.3) (1.11.3)
Collecting pybgcode-jneilliii (from BGCode==0.2.3)
Downloading pybgcode_jneilliii-0.2.1-cp311-cp311-manylinux_2_26_armv7l.manylinux_2_31_armv7l.whl.metadata (2.5 kB)
Requirement already satisfied: OctoPrint-FileCheck>=2024.11.12 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2025.7.23)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2025.5.14 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2025.7.23)
Requirement already satisfied: OctoPrint-PiSupport>=2023.10.10 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2025.7.23)
Requirement already satisfied: argon2-cffi>=23.1.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (23.1.0)
Requirement already satisfied: Babel<2.17,>=2.16 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2.16.0)
Requirement already satisfied: cachelib<0.14,>=0.13.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (0.13.0)
Requirement already satisfied: Click!=8.2.0,<9,>=8.1.8 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (8.1.8)
Requirement already satisfied: colorlog<7,>=6.9.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (6.9.0)
Requirement already satisfied: emoji<3,>=2.14.1 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2.14.1)
Requirement already satisfied: feedparser<7,>=6.0.11 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (6.0.11)
Requirement already satisfied: filetype<2,>=1.2.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (1.2.0)
Requirement already satisfied: Flask-Assets<3,>=2.1.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2.1.0)
Requirement already satisfied: Flask-Babel<5,>=4.0.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (4.0.0)
Requirement already satisfied: Flask-Login<0.7,>=0.6.3 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (0.6.3)
Requirement already satisfied: Flask-Limiter<3.9,>=3.8 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (3.8.0)
Requirement already satisfied: flask<3.1,>=3.0.3 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (3.0.3)
Requirement already satisfied: frozendict<3,>=2.4.6 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2.4.6)
Requirement already satisfied: limits<5 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (4.1)
Requirement already satisfied: markdown<3.8,>=3.7 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (3.7)
Requirement already satisfied: netaddr<1.4,>=1.3.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (1.3.0)
Requirement already satisfied: netifaces<0.12,>=0.11.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (0.11.0)
Requirement already satisfied: packaging in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (24.2)
Requirement already satisfied: libpass<2,>=1.8.1 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (1.9.1.post0)
Requirement already satisfied: pathvalidate<4,>=3.2.1 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (3.3.1)
Requirement already satisfied: psutil==6.0.0 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (6.0.0)
Requirement already satisfied: pydantic<3,>=2.10.6 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2.11.7)
Requirement already satisfied: pylru<2,>=1.2.1 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (1.2.1)
Requirement already satisfied: pyserial<4,>=3.5 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (3.5)
Requirement already satisfied: pytz in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2025.1)
Requirement already satisfied: PyYAML<7,>=6.0.2 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (6.0.2)
Requirement already satisfied: requests<3,>=2.32.3 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (2.32.3)
Requirement already satisfied: sarge==0.1.7.post1 in /opt/octopi/oprint/lib/python3.11/site-packages (from OctoPrint->BGCode==0.2.3) (0.1.7.post1)
[... cut ...]
Building wheel for BGCode (setup.py): finished with status 'done'
Created wheel for BGCode: filename=BGCode-0.2.3-py2.py3-none-any.whl size=4435 sha256=d155131c3452b3e60a1c5505a88d6252461fdb608fc6a928bb73ca0abb2dc38f
Stored in directory: /tmp/pip-ephem-wheel-cache-01fkqx70/wheels/ee/db/6f/e30ee50e4e22ed82284f63581259579a467e0c42d2a9e53e51
Successfully built BGCode
Installing collected packages: pybgcode-jneilliii, BGCode
Successfully installed BGCode-0.2.3 pybgcode-jneilliii-0.2.1
Done!
```

Thanks @jneilliii !

jneilliii, 2025-09-17:
Yep, that's a good install, should work with that file format now.

closed 2025-09-17 by jneilliii