jneilliii/OctoPrint-BGCode #15: .bgcode leftover files never being deleted

redRock14, 2025-05-29:
There seems to be an issue with the translation process where a .bgcode files are being saved in the _/home/[user]/.octoprint/data/bgcode_ folder and never deleted.  After time, and many files accumulating in folder the result is high disc usage.

Originally reported as a question in the Octoprint Forum and that can be found here:  
https://community.octoprint.org/t/previously-printed-bgcode-files/63506

In that thread, you requested for me to create this issue request.  

thanks!

redRock14, 2025-05-29:
OK that log file is not going to help much; logging level was not set to DEBUG.  I set error loging to DEBUG and this time I can see the log entry at line 31:

[image]

I don't see the log entry thrown by the exception handler though.

jneilliii, 2025-05-29:
Yeah, no obvious errors there at all. I'll do some experimenting this weekend.

redRock14, 2025-06-24:
@jneilliii - I haven't been able to dig more into this.  I finally got my Core One upgrade kit, and currently I'm working on putting it all back together.  

But, before I started the upgrade I just wrote a bash script to clear the folder out.  Then set it up with a cron expression to delete the files once a day.  Once I get the printer back together I would be happy to help figure this out if you like.  Just let me know if you want me to try something in particular.

Mercur13, 2025-06-24:
@jneilliii - here a small exerpt of the \\\bgcode-Folder (totally over 400 Files):

[image]

I would like to assist, what do you need else ?

redRock14, 2025-06-24:
@jneilliii - I uploaded a couple of files.  Both sliced with PrusaSlicer.  One as a CoreOne, the other as a MK4 compatible printer files.  Both were gcode files:

[image]

jneilliii, 2025-06-24:
I released an updated version that includes additional debug logging. If you guys can make sure you're on version 0.2.1 and in OctoPrint's logging section set the plugin to debug level and upload/send a file to OctoPrint and then share your octoprint.log here that could be helpful.

redRock14, 2025-06-24:
@jneilliii - does this help?  Deleted the log file, then restarted OctoPrint, then uploaded a file.

[octoprint.log](https://github.com/user-attachments/files/20885335/octoprint.log)

jneilliii, 2025-06-24:
Thanks guys, so the underlying issue is something to do with the filewrapper with move=True defined. OctoPrint should be handling the process for that gcode file, but for some reason that is not happening. I will have to debug this further with @foosel I think.

jneilliii, 2025-06-24:
@redRock14 I just noticed something in your log. Does the same thing happen if you disable the dashboard plugin?

redRock14, 2025-06-24:
@jneilliii - That is it!  I just disabled the Dashboard plugin....

[image]

Uploaded a file...  The same one from before.  And the folder is empty...

[image]

And here is the entry in the log...

[image]

So I suppose the Dashboard plugin has some sort of issue then?

jneilliii, 2025-06-24:
I just happened to notice in your log that it also seems to include a file preprocessor, so not sure how that is handled in core OctoPrint. I know I set the order of mine to run as early possible for all registered preprocessors.

jneilliii, 2025-06-26:
> the time required for compression in the slicer and decompression in Octoprint will cause a noticeable delay for large files (up to 255 MB in my case), right?

It's a trade-off one way or another, either processing time to compress/decompress in order to upload smaller files or upload time of the larger uncompressed files. Honestly, on a local network sending files from the slicer in standard gcode format has never been an issue for me personally in the 10+ years I've used OctoPrint.